UAA-Robo/CoffeeBot #13: Research Vision Cameras

TonyP820, 2023-02-12:
Research various vision cameras; select 3 viable solutions with the following budget considerations. (unlimited budget, reasonable budget, dumpster diving budget).
You will own whichever solution is ultimately decided upon and will be expected to become the "resident expert" upon acquisition and arrival.
Multiple levels are both for team consideration and investor consideration during our pitch.

patpragman, 2023-02-14:
**Solution 1:**  I just bought a couple of these [cameras](https://www.amazon.com/gp/product/B084ZJFNKN/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1)

They're cheap and they work pretty well.  I used them on the BroPro (ask Jaren).  I have to prepare for a presentation but I can provide more camera stuff later.

patpragman, 2023-03-05:
@TonyP820 - do we have a selection for this, or are we still just kind of unsure?  I know the scope of the miniature changed a bit.

TonyP820, 2023-03-05:
@patpragman No, we don't have a selection as of yet; we aren't using any cameras for the scaled-down version.

closed 2023-08-26 by AnthonyvW