UAA-Robo/CoffeeBot #20: Claw CAD

micahkayle, 2023-02-18:


DawsonTP, 2023-02-19:
Claw “fingers” CAD basics done, will refine for more detail later
Basic baseplate done
Todo:
Must investigate motor mounting (remove material from finger cogs to make room for stand-offs?)
- Redesign baseplate for this use plus ready effector for rotational mount to larger motor for 270 degree spin

Dawson-Packer, 2023-02-21:
Claw CAD expected to have first iteration done by Friday meeting (2/24/23). Much progress made on claw assembly. See CoffeeBot Log for more details.
Todo:
- Place motor in CAD
- Bore hole in pivot plate for swivel motor
- Split drive finger into two fingers

Dawson-Packer, 2023-02-22:
Fixed Pivot plate dimensions. Removed material from top and base plate to make claw lighter. Detailed fingers. Driving finger has two contact points. Driven finger has one.
Todo:
- Place motor in CAD
- Bore hole in pivot plate for swivel motor
- Look at additional support for motor.

Dawson-Packer, 2023-02-25:
First actual component 3D-printed yesterday. Will continue to print on Monday. Look at closing this task when full model is functional.

Dawson-Packer, 2023-02-27:
Interfacing grooves on printed part don’t fit. Must revise (Monday)

Dawson-Packer, 2023-03-05:
Interface fixed! Everything on schedule now.

Dawson-Packer, 2023-04-22:
First Iteration done

closed 2023-04-22 by Dawson-Packer