UAA-Robo/CoffeeBot #36: Research TTS that can run locally

AnthonyvW, 2023-08-26:


epicexcelsior, 2023-09-27:
[PiperTTS](https://github.com/rhasspy/piper) seems promising. [Here](https://discord.com/channels/615798930952683520/1141783974298013706/1150941439841280111) is a sample on Discord. It was able to generate the voices in under 3 seconds fairly consistently, which can be improved using live audio playback (play audio as it's generated) and leaving the process running.
The voices don't sound too bad but their not very good either. I'll look into ways to improve that (e.g. finding better models or training our own) or see if there's a high quality option that doesn't sacrifice speed.

epicexcelsior, 2023-10-04:
Currently looking at [Coqui TTS](https://github.com/coqui-ai/TTS), Piper (above), and [Tortoise](https://github.com/neonbjb/tortoise-tts). I've heard that Tortoise is very realistic and high-quality, but also insanely slow. Coqui hasn't been tested yet, but I know it doesn't natively support raw/real-time audio playback/streaming (playing audio as it's generated) as Piper can do.

Since the Coffeebot's main purpose is to facilitate coffee ordering & serving, it might be a good idea to pre-generate the most frequently used phrases, store them as audio files, and then just play them. Most interactions could fit into a predictable flowchart. Other phrases could be generated with Coqui or Piper. This would allow high-quality voices and remove the generation wait time at the cost of flexibility.

Thoughts?